mhostetter/gr-adsb #2: Static filename

theforcedk, 2016-04-13:
You're writing to a specific filename in:

<pre>
./python/decoder.py:        self.fp_csv = open("/home/matt/adsb.csv", "a")
</pre>

mhostetter, 2016-04-13:
Yes, definitely!

theforcedk, 2016-04-13:
@mhostetter Is it worth adding a low pass/fir filter, or is it better to just grab all the signal and try and process it?

mhostetter, 2016-04-13:
You're right.  That would probably be beneficial because it would kill the out-of-band noise.  I'm trying to write the code so it will support arbitrary sampling rates.  I'm using a HackRF, so I'd like to support 2, 4, 8, 10, 16, 20 Msps (if the CPU can handle it!).  So for the higher sampling rates, an FIR LPF would be beneficial I think.  The question is what bandwidth filter.  I don't want to filter the signal at all.  I think the filter would need to be at least 2 MHz wide, but maybe slightly more.  I'll play around with adding that!  I'm sure the performance will improve.

mhostetter, 2016-04-13:
What I'd really like to do is configure the HackRF in the osmocom source to reduce the ADC filter bandwidth.  So I'm sampling at 4 Msps but I'd set the ADC anti-alias filter to ~2 MHz.  That's a "free" filter that accomplishes the same goal.  I don't think gr-osmosdr supports this though.  Might have to add it!

theforcedk, 2016-04-13:
@mhostetter That's actually a really nice idea, using ADC directly on the HackRF. It should be a bit faster than doing it in software (gnuradio), altho it's also doing it in software on the HackRF, but it should be faster as it's not hammering the USB with unnecessary traffic.

I'll play around with that too :)

theforcedk, 2016-04-13:
@mhostetter Not the best in gnuradio just yet. I managed to get dump1090 working using hackrf_transfer. We can most likely test the idea using this - the hackrf tools are also a bit easier to modify than gr-osmosdr (faster). You can test it by doing:

**Please be aware of that I activated bias-t (antenna power) as I'm using a LNA. This is the "-p 1" switch. Remove if it you don't need.**

<pre>
hackrf_transfer -r 1090.sb -f 1090000000 -s 2000000 -p 1 -a 1 -l 37 -g 47
</pre>

HackRF spits out 8 bit signed I/Q samples, but dump1090 expects unsigned, so convert it using sox to signed (1 channel) - there might be a stdin option that can handle it in one go.

<pre>
sox -r 2000000 -c 1 1090.sb 1090.ub
</pre>

Then load up dump1090

<pre>
./dump1090 --ifile /Users/yoda/1090.ub --interactive
</pre>

[image]

mhostetter, 2016-04-13:
Ok, that's cool.  I'll have to play with it.  I know about dump1090, but I thought it would be fun to write my own decoder.

I found the antenna to be very important.  I have this one, but it appears to be out of stock.  It has positive gain (in dBi) at 1090 MHz.
[http://www.amazon.com/dp/B00WZL6WPO?tag=fligh01-20](url)

theforcedk, 2016-04-13:
Sure, I'm all in in regards to working on the decoder :) - testing output from HackRF is just a bit easier in hackrf_transfer for me, as I'm not that well versed in gnuradio/python. I work as a software developer, so the programming aspect is not new to me, the whole radio thing is.

Great tip on the antenna, I'll have to get me one of those.

Something like this should work for a headless test (eg. no files). Disk would be filled way too fast with a 2Mhz samp rate on HackRF dumping it to a file.

<pre>
#!/bin/sh
hackrf_transfer -r /dev/stdout -f 1090000000 -s 2000000 -p 1 -a 1 -l 40 -g 48 | sox -t ub -r 2000000 -c 1 /dev/stdin -t sb - | /Users/yoda/Development/Source/dump1090/dump1090 --aggressive --raw --ifile -
</pre>

But currently trying to get it working.. something is not getting outputtet correctly.

Looking forward to watching the progress on this repo :)

mhostetter, 2016-04-17:
I added code to enable/disable CSV logging and configure the file name from the GNU Radio Companion GUI.

commit b90fd7bd (2016-04-17): Closes #2.  Added code to enable/disable CSV logging and configuring the CSV filename from the GRC GUI.  Also added placeholder code for configuring database logging.
closed 2016-04-17 by mhostetter