mhostetter/gr-adsb #11: Setting PKG_CONFIG_PATH for cppunit doesn't work.

romeojulietthotel, 2017-03-03:
The cmake stage fails because cmake does not find cppunit include files even with PKG_CONFIG_PATH set properly. The cppunit is in a non-standard location. I worked around it by adding the paths to cmake/Modules/FindCppUnit.cmake.

If I run
`PKG_CONFIG_PATH=/here/is/cppunit/lib/pkgconfig pkg-config --cflags cppunit`

or

`PKG_CONFIG_PATH=/here/is/cppunit/lib/pkgconfig pkg-config --libs cppunit
`
I get correct results.

mhostetter, 2017-03-10:
Thanks for the feedback.  What is your system configuration?  I'm using the default cmake file from GNU Radio's gr_modtool.  Do you have GNU Radio already installed?

romeojulietthotel, 2017-03-18:
I guess gnuradio is forcing cppunit on modules but it does not use it.

mhostetter, 2017-03-31:
Do you have a suggestion for what I should do? I think the CMake will work fine if GNU Radio is installed with PyBOMBS. I think it covers all the dependencies.

romeojulietthotel, 2017-03-31:
Ok, then I think this is a documentation bug or I missed that PyBOMBS was a req.

closed 2017-03-31 by romeojulietthotel

mhostetter, 2017-03-31:
Well, PyBOMBS isn't a requirement necessarily. But PyBOMBS is a nice way to install GNU Radio, ensuring all dependencies are satisfied.

Are you still having trouble with this? Does it prevent the module from building?