mhostetter/gr-adsb #12: gr-adsb

Eivindo, 2017-03-30:
Hello. Your gnuradio gr-adsb setup, will it work with RTL-SDR?

mhostetter, 2017-03-30:
Yes, it does work with RTL-SDR through the Osmocom source. Just set the center frequency to 1090 MHz and the sample rate to 2 Msps.

I'm going to add a USRP/Osmocom source option to the example flowgraph tonight.

Let me know if you have trouble building/using the gr-adsb module.

closed 2017-03-30 by mhostetter

Eivindo, 2017-03-31:
Thanks for your reply.

We have a Nuand bladeRF, Ettus B200, and RTL-SDR dongle. We got your design
to work great with the RTL-SDR, but when we connected the Ettus B200 and
bladeRF, we received a signal, but could not decode he ADS-B signal (see
attached image).

Could you send us config for your USRP block, or any other pointers?

We are trying to use this as part of our College Bachelor thesis project.

Would it be too much to ask for you to draw us a signal schematic of this
Gnu radio setup? We are a bit unsure of what happens in the blocks you have
made.

Thanks in advance!   :)

Eivind Andresen
University College of South-east Norway

[image: Innebygd bilde 1]).

mhostetter, 2017-03-31:
I couldn't view the attached image.

Are you using the example flowgraph I included here https://github.com/mhostetter/gr-adsb/blob/master/examples/adsb_rx.grc?

What is the center frequency and sample rate for the B200? I also use a B200 with no problem.

mhostetter, 2017-03-31:
I would recommend using the example flowgraph. Make sure you connect the antenna to the correct port. The example flowgraph is configured for the `TX/RX` port. You can use `RX2` if you want, but you'll need to change the USRP source block.