mhostetter/gr-adsb #13: GNU Radio printing

gornickijakub, 2017-05-15:
Hello,

I'm trying to use your projekt with ZC706 SDR board and FMCOMMS2 RF module, I configured Fmcomms2/3/4 GNU Radio block so it has LO set at 1090 MHz, bandwith set to 50 KHz and sampling rate to 2,084 MSPS. When I run the application in GNU Radio everything seems to work correctly, amplitude sink shows some data and frequency sink I added shows the badnwith correctly as well. However, I'm not sure where the data should be printed from decoded ADS-B signals. Does ZMQ PUB Message Sink should print it in console or it will be written only to adsb.db database? I checked the the database after receiving data and it appears to be empty. I suppose my configuration is wrong, but I'm not sure where should I look for results.

Regards,
James

mhostetter, 2017-05-15:
Thanks for using the repo! Here are some thoughts.

> I'm trying to use your projekt with ZC706 SDR board and FMCOMMS2 RF module, I configured Fmcomms2/3/4 GNU Radio block so it has LO set at 1090 MHz, bandwith set to 50 KHz and sampling rate to 2,084 MSPS. 

I'm not familiar with this SDR, but your SDR parameters should be:

- Center frequency should be 1090 MHz, which it appears you have.
- Sample rate should be a multiple of 2 Msps. You wrote 2084 Msps, but I'm assuming you meant 2.048 Msps. If you can change it to 2.000 Msps that would be better, but if not you can probably get away with it. The demod/decoder is expecting an integer number of samples per pulse, where each pulse is 0.5 us (i.e. N*2.0 Msps).
- The bandwidth should be equal to the sample rate. By bandwidth, I assume you mean the bandwidth of the anti-alias filter before ADC sampling. The signal occupies a bandwidth of 2 MHz, so you're bandwidth should be 2 MHz, not 50 kHz. I suspect this is giving you most of your problems.

> When I run the application in GNU Radio everything seems to work correctly, amplitude sink shows some data and frequency sink I added shows the badnwith correctly as well. However, I'm not sure where the data should be printed from decoded ADS-B signals.

Can you attach some screenshots? Make sure the "Burst Threshold" is low enough that you can decode weak signals, but not too low that it's triggering on noise. You can change it from the GUI to find the optimal value for your SDR/gain combo. As far as data print outs, set the `ADS-B Decoder`'s `Print Level` to `Brief` or `Verbose`. This will print the decoded packets to `stdout`.

> Does ZMQ PUB Message Sink should print it in console or it will be written only to adsb.db database? I checked the the database after receiving data and it appears to be empty. I suppose my configuration is wrong, but I'm not sure where should I look for results.

The ZMQ Pub Message Sink sends the decoded data to the `webserver.py` script (that has to be run separately) that publishes the data on a Google Map on `localhost:5000`. There are additional dependencies for the webserver, though. I talk about them on the README page. I don't think I have the database logger in the example flowgraph currently. I am migrating to using another OOT module I have called [gr-sqlite](https://github.com/mhostetter/gr-sqlite). I can push up those changes if you're interested in logging to a SQLite database as well. But you should check out the webserver, it's pretty fun.

Let me know if you can (or can't) get it working!

Matt

gornickijakub, 2017-05-16:
> Center frequency should be 1090 MHz, which it appears you have.
> Sample rate should be a multiple of 2 Msps. You wrote 2084 Msps, but I'm assuming you meant 2.048 >Msps. If you can change it to 2.000 [...]

Yes I made a typo there, sampling frequency is 2.084 Msps, Fmcomms2 doesn't allow to input lower value so I can't set it at 2.000 Msps. I corrected the badnwith so it's the same as sampling frequency now. Fmcomms2 receiver in gnu radio takes shorts as input, I suppose correct conversion value may be very important here. I tried with 1, 4, 16, 32, 64, 128 values and 64 provides the highest frequency of changes in time domain scope, so I left it like that for now.

Seems like there's still problem with my configuration, I attach the screenshot of workspace and two scopes I'm monitoring. I run webserver and changed print level to verbose but haven't got any output data yet. I'll try to find the best burst threshold value after you look at the blocks configuration, I have a hunch something is still wrong there. 

[image]
[image]

mhostetter, 2017-05-16:
I think you're getting close.

> Fmcomms2 receiver in gnu radio takes shorts as input, I suppose correct conversion value may be very important here. I tried with 1, 4, 16, 32, 64, 128 values and 64 provides the highest frequency of changes in time domain scope, so I left it like that for now.

As far as scaling the short outputs, I would scale by 2^15 so the output is limited to +/- 1. Also, I assume the first yellow short output is the real and the second the imaginary?

What antenna are you using? Any LNA or band-pass filter?

I think you might not have specified the sample rate in the Frequency Sink, as it only appears to have 1 MHz of bandwidth. Something to check.

Also, I would a Waterfall Sink. It's easier to see the bursts in the waterfall. You should expect something like this (if you have a good antenna, RF front end, etc, etc). This is a screenshot from early in the development, so the `stdout` prints have changed and the Time Sink plot a little too. Also, I was using 4.0 Msps for this screenshot.
[image]

mhostetter, 2017-05-16:
I'm starting to think the sample rate offset is really hosing you. Can you increase the rate to 4.000 Msps? If not, you might need a rational resampler to convert 2.048 Msps to 2.000 Msps in DSP. Look into to GNU Radio's `Polyphase Rational Resampler` block. Your rsampling rate would be `2.000/2.048`.

gornickijakub, 2017-05-17:
>As far as scaling the short outputs, I would scale by 2^15 so the output is limited to +/- 1. Also, I assume >the first yellow short output is the real and the second the imaginary?
That's right, board receives IQ samples with two channels. I've read a few threads in Analog Devices forum and found out I need to scale channels by 4096 to limit output to +1 or -1. 

I'm using LNA antenna, I configured GUI sinks with correct bandwith and finally changed sample rate to 4 Msps (from previous research I know it can handle sample rate up to 12.5 Msps without issues). After setting it up as you adviced it finally works!

I managed to receive signal from an airplane arriving to local airport. I also checked webserver, it runs but doesn't show detected planes - I suppose it's because I connect GNU Radio with SDR through Ethernet, I noticed I need internet connection to load up map so it's possible it can't trace flights without access to location data. It's not a big deal, as long as I can receive ADS-B signal and decode plane's information I'm content. 

I'm very grateful for help and providing an implementation of this receiver. I attach screens of final configuration, scopes and decoded data in console. Thank you very much!

Regards,
James

[image]
[image]

closed 2017-05-17 by gornickijakub

mhostetter, 2017-05-17:
Awesome! That's looking great. It's fun, right?

As far as the webpage is concerned, it's true that it fetches the map background from Google Maps at page launch. So if you're on a LAN w/o access to the internet, that probably won't work. You might be able to access the Internet through WiFi while connected to your SDR via Ethernet.

Happy decoding!