mhostetter/gr-adsb #14: gr-adsb question

kynszbd1, 2017-11-27:
Hi, thank u for your demo. I have a question, I run the demo of the "webserver.py" ,but when I open a web broswer,and go to localhost:5000, I cannot get any information. I wonder if there are any other step or configuration or demand on my computer. Thank you very much and look forward to your reply  :)

mhostetter, 2017-11-27:
For the webserver to work, you'll need to install the [Webserver Dependencies](https://github.com/mhostetter/gr-adsb#webserver-dependencies). Just check to make sure you have those.

Do you see the Google Maps page load when you go to `localhost:5000`? Do you just not see any planes?

Are you running the example flowgraph in the background? Is the ZMQ Pub Message Sink enabled? Can you set the print level on the ADS-B decoder to "Brief"? You should see stdout prints if its decoding. Then we can debug why they're not showing up on the map.

kynszbd1, 2017-11-28:
thank your reply.I have installed the Webserver Dependencies.I don't see the Google Maps page load when  I go to localhost:5000.I  run the example flowgraph in the background, but i see the  flowgraph  has  some problems, there is a" missing block",which colour is red . then,there is some errors in the terminal like these "File "/usr/lib/python2.7/posixpath.py", line 73, in join
    path += '/' + b
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe4 in position 11: ordinal not in range(128)".
I think the ZMQ Pub Message Sink  disabled,I can not  set the print level on the ADS-B decoder to "Brief".Therefore,I can not see stdout prints . Thank you very much and look forward to your reply .

mhostetter, 2017-11-28:
- Internet access is required to load the Google Maps page, so check that.
- Which block is red ("missing block") in the flowgraph?
- If the ZMQ Pub Message Sink is disabled, then re-enable it by clicking on the block and clicking `CTRL+E`.
- Why can't you change the decoder print level to "Brief"?

Also, can you attach a screenshot of your GNU Radio Companion screen?

kynszbd1, 2017-11-29:
can you give me your e-mail ,there is not very convenient to place the picture.
[image],
[image].There is the problem I have encountered.

mhostetter, 2017-11-29:
Ok, you can delete the `sqlite_sink` block. It is not needed. Can you also copy and paste all of the stdout error message (it got cut off). It looks like your having a python flask issue.

Does the flowgraph run? What is its output?

mhostetter, 2019-07-20:
I'm sorry I never got back to you. Were you able to resolve this issue? I'm going to close this, just due to the elapsed time. If you are still experiencing this issue (and are still using the software), please re-open it.

closed 2019-07-20 by mhostetter