mhostetter/gr-adsb #16: how to debug your project

shianglin, 2018-11-22:
Hi~ Thank you for your demo.  I am a beginner. I am trying to run your demo with pycharm and trying to debug how it works. Firstly, I have already installed this project. Then, I created the "adsb_rx" python script using gnuradio-companion. I ran the adsb_rx.py in pycharm, as shown in below.
[image]
[image]

Then, I set a breakpoint at "def work()"of framer.py, as shown in below.
[image]

When I ran with debug mode. The program doesn't stop at my breakpoint. I didn't know why. Could you please tell me how did you debug your program?

 Thank you very much and look forward to your reply :)

mhostetter, 2018-11-26:
Sorry for the late reply. I have never debugged Python in GNU Radio before, although I probably should.

This is what I think is going on. GNU Radio installs the Python scripts/modules into either `/usr/local/lib/...` or your target prefix when installing with PyBOMBS. So the break point you set in your local `framer.py` is the not the `framer.py` that is actually called when running `adsb_rx.py`, the installed version is. One quick way to test this is to put the break point inside `adsb_rx.py` and then step into the top block's `run()` method.

Alternatively, you can find the install directory of `framer.py` and put a break point in that file in PyCharm.

Let me know if it works.

shianglin, 2018-11-27:
Hi~ Thank you for your reply. 
Actually, you said the first way that I have already tried. It still didn't stop at def work() function of framer.py when I ran the debug mode.
The second way that I also tried. All the python codes of ads-b module have installed in here "/usr/local/lib/python2.7/dist-packages". I  put a breakpoint in that file where I installed, but it still didn't stop at def work() function of framer.py when I ran the debug mode. Although I think it should be worked.

I tried to find the other way on the website. All information for debug gnuradio block used "gdb", such as "https://wiki.gnuradio.org/index.php/TutorialsDebugging". However, this demo couldn't use gdb due to its python code. So far... I have no idea how to debug this program with Pychram compiler.
if you have any ideas plz kindly let me know.
Thank you very much and look forward to your reply :)

mhostetter, 2018-12-11:
Sorry for the late reply. Unfortunately I don't have any other ideas. I would've thought one of those two methods would've worked. Apologies that I don't have a better (any) solution. Maybe post on the GNU Radio listserv to see if anyone has ideas.

closed 2018-12-11 by mhostetter