mhostetter/gr-adsb #17: No module called adsb

corebloom, 2019-03-03:
Hello Matt

Thank you very much for your project. Im trying to get it to run with a hackrf.

I have manually built your code from source. I see the new Adsb-blocks in Gnuradio (no longer red when i open your example flowchart).

However when i want to execute the flowchart it stops on line 33 in the .py script with the error message that he cannot load the module adsb because there is no such module.

What am I doing wrong? Btw i am running this on pentoo. 

Many thanks and cheers
corebloom

corebloom, 2019-03-03:
This is the exact message im getting from gnuradio:

..adsb_rx.py", line 33, in <module> import adsb
ImportError: No module named adsb

>>> Done (return code 1)

closed 2019-03-03 by corebloom

mhostetter, 2019-03-03:
Were you able to resolve this issue. I have some thoughts on what the issue might be, if it still remains.

corebloom, 2019-03-03:
Hello Matt

Thank you for your message. I think i actually was yes :) Im pretty new to
linux, so trial and error on my side.

 I manually copied a folder named adsb from /lib/local/lib64/python2.7 to
the same folder my flowgraph grc file is stored. it now shoes recieving
data! so nice.

If im already in contact with you, thanks again for your code, really cool
stuff.

2 questions:

is it possible to outout the sdout data to some other place then the output
window in gnuradio?

do you think i should install a 1090mhz filter inbetween the hackrf and the
antenna? i have a flightaware filter lying around.

Thanks so much

mhostetter, 2019-03-04:
My guess as to why python failed on `import adsb` is that the `gr-adsb` project was built but installed to the wrong directory (as compared to where GNU Radio is expecting OOT modules). How did you install GNU Radio? With PyBOMBS? Did it come pre-installed in your distro?

If that is the issue, the way to resolve it is to specify `-DCMAKE_INSTALL_PREFIX=<path-to-gnu-radio-install-location>`. If using PyBOMBs, you likely specified a target prefix. If GNU Radio was pre-installed in your distro, it may have been installed in `/opt/`.

As to your questions:
1. You can redirect the stdout prints to a file, for instance, using the Linux command `tee`. You may need to install `tee` with your distro's package manager. The `adsb_rx.grc` flowgraph can be (and in fact always is) built by GNU Radio. It is "compiled" into adsb_rx.py. This script can be run from the command line, without using GNU Radio Companion. So to "tee" the output, you'd run `$ python adsb_rx.py | tee output.log`.
2. Yes, I would recommend a filter between the antenna and the HackRF. I have the same antenna. I previously used a HackRF to develop this project. And it receives well. I later upgraded to a USRP B210, and have to admit it has better reception/range. The 8-bit IQ sample resolution can be a little limiting with HackRF.

corebloom, 2019-03-04:
Thanks a lot for your detailled response. 
You definitly got my issue correctly: gnuradio came preinstalled with my distro, and i definitly didnt cmake gr-adsb in the correct directory. However apparantly i fixed the problem with manuall moving the folder as described in my post above.
However i learnt something for the future, thank you!

Also thanks a lot for your response on the tee command, that was exactly what i was looking for. Perfect.

Ah i just took a look at the USRP B210, looks very good but unfortunatly also a bit pricey :) I have enough to plwlay with at the moment with my HackRF, but might be a consideration in the future!

Thanks again Matt