mhostetter/gr-adsb #18: Minor documentation update

potto216, 2019-07-20:
First off, thanks for writing the software! It works well with my RTL-SDR. If the project is still being maintained I can create a pull request to update the readme with some suggestions. They would be:
1. Add pip install zmq and pmt
2. To prevent accidentally messing up the user's root python environment have optional instructions to use pipenv.

mhostetter, 2019-07-20:
You're very welcome. I'm happy to hear it's working and being used :smile: 

While I haven't directly developed on this project in a couple years, I still track/maintain it. Also, I have plans to update the code (both demodulator and ADS-B decoder) when I upgrade this project to GNU Radio 3.8. I would absolutely welcome a pull request. Thanks for using the software.

potto216, 2019-07-20:
Great! Will do, yeah moving over to 3.8 will be interesting with the Python 3 and YAML changes.

potto216, 2019-07-21:
If you open a new issue for updating the demodulator I would like to help. From what I've read ADS-B is basically transmitting a form of on off keying (which I found shocking) that explains why you are doing incoherent demod. Can you improve the performance with coherent demod? I did a quick Google Scholar search and found a few articles, but I'm not sure what the constraints are.

mhostetter, 2019-07-22:
I like the demod conversation. I'm going to move that conversation to #21 and close this issue since you already submitted PR #20.

closed 2019-07-22 by mhostetter