mhostetter/gr-adsb #19: Update Google Map Server code/instructions

potto216, 2019-07-20:
I'm having problems using the Google Map Server example. It is warning me that the keyless option will soon be deprecated and what is worse I cannot get the detected tracks to show up. If the project is still being maintained I would be happy to debug the issue and submit a PR.

mhostetter, 2019-07-20:
Help here would be very welcome. A while back I did notice a similar issue. At the time of development, operation worked well without an API key. I think users received N free uses (tile queries, etc) per day. It seems Google is cracking down a bit.

mhostetter, 2019-07-20:
Also, full disclosure, this was my first foray into web development of any kind. So, the javascript and Flask webserver surely could be improved. :laughing:

potto216, 2019-07-20:
Thanks for the info, I was planning on forking the repo and creating a branch PR unless you rather I make the changes on master.

mhostetter, 2019-07-20:
Yes, that sounds good. I would make the changes on a separate branch.

mhostetter, 2019-07-23:
I am testing now with my setup. The Google Maps webpage gives me "For development purposes only" watermarks and a `Google Maps JavaScript API warning: NoApiKeys https://developers.google.com/maps/documentation/javascript/error-messages#no-api-keys` warning, but I do see the plane tracks.

Can you provide me any more diagnostics (webserver stdout, webpage console, etc) on your setup. I might be able to help you debug why you aren't seeing planes on the webpage.

potto216, 2019-07-23:
I was getting
```
(ads_b_app) user@sdr:~/src/ads_b_app/web$ python ./webserver.py
Waiting for zmq socket to connect.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/user/.pyenv/versions/3.7.4/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/home/user/.pyenv/versions/3.7.4/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "./webserver.py", line 42, in background_thread
    socket.setsockopt(zmq.SUBSCRIBE, "")
  File "zmq/backend/cython/socket.pyx", line 426, in zmq.backend.cython.socket.Socket.set
TypeError: unicode not allowed, use setsockopt_string

 * Restarting with stat
 * Debugger is active!
Waiting for zmq socket to connect.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/user/.pyenv/versions/3.7.4/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/home/user/.pyenv/versions/3.7.4/lib/python3.7/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/user/src/ads_b_app/web/webserver.py", line 42, in background_thread
    socket.setsockopt(zmq.SUBSCRIBE, "")
  File "zmq/backend/cython/socket.pyx", line 426, in zmq.backend.cython.socket.Socket.set
TypeError: unicode not allowed, use setsockopt_string
```

Which went away when I forced a binary string in 
`socket.setsockopt(zmq.SUBSCRIBE, b"")`
in webserver.py. I'll try it with live data shortly and add a comment with my result.

mhostetter, 2019-07-23:
The python webserver needs to run with Python 2.7. I probably should make that more clear in the docs. It's very likely the `zmq` and `flask` packages are different for Python 3. It appears your `python` symlink is referencing Python 3.7.

potto216, 2019-07-23:
Oh okay my bad, I'll try it with Python 2.7. That will be an easy issue to close:)

potto216, 2019-07-24:
I was able to get the map server code to work without problems when I had properly set my environment to Python 2.7. Therefore this is not an issue an I am closing it.

closed 2019-07-24 by potto216