mhostetter/gr-adsb #21: Demod improvements

mhostetter, 2019-07-22:
Opening this thread to continue the discussion from #18.

potto216, 2019-07-23:
@mhostetter Thanks for the explanation and the link. I can see your point especially because you can run the algorithms in python. Is the objective of this issue to first convert and test everything with python 3?

mhostetter, 2019-07-23:
Python 3 doesn't have to come first. Although, your interest in the project has rekindled mine. I'm working on installing/running GNU Radio 3.8 now. And will start porting/upgrading code soon.

If you have other demod ideas or implementations, feel free to start experimenting. It should be easy enough to integrate, regardless of where I am in the Python 2 to 3 conversion.

potto216, 2019-07-24:
Will do. At the moment I can't think of any improvements to the demod while I'm learning ADS-B. But I'll let you know if I come up with something.