mhostetter/gr-adsb #22: Timestamp

Tyoluyi, 2019-07-22:
Excuse me, could I ask how to export the timestamp info for each message through your flow chart?    Thank you

mhostetter, 2019-07-23:
Can you elaborate on exactly what you'd like?

If you just want the timestamps to print for each message, you can connect a Message Debug block to the ADS-B Decoder's `decoded` output. You should also set Print Level to None on the decoder. See below:

[image]

```
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
((vertical_rate . -128) (speed . 418.403) (datetime . 2019-07-23 02:10:38 UTC)
(latitude . 38.9556) (snr . 22.1123) (callsign . ENY3980 ) (icao . abb282)
(heading . -137.809) (longitude . -77.4204) (altitude . 36000) (timestamp . 1563847838)
(df . 17) (num_msgs . 12))
pdu_length = 112
contents = 
0000: 01 00 00 00 01 01 00 01 01 00 01 00 01 00 01 01 
0010: 01 00 01 01 00 00 01 00 01 00 00 00 00 00 01 00 
0020: 01 00 00 01 01 00 00 01 00 00 00 00 01 01 00 01 
0030: 00 00 01 01 00 01 01 01 01 00 01 00 00 00 01 01 
0040: 00 01 00 01 01 00 00 00 00 00 00 00 01 01 00 00 
0050: 00 01 00 00 01 00 00 00 01 00 01 00 01 01 01 00 
0060: 01 00 00 00 01 01 00 00 00 01 00 01 00 01 01 00 
***********************************
```

If you would like the timestamps/datetimes to display in Verbose printing mode, that's something that could be easily added.

Tyoluyi, 2019-07-23:
Dear mhostetter
         Hi, I’m a graduate student and I’m working on ADSB and MLAT based
on SDR. I’m trying to get the timestamp for each message and then send the
TOA data & ICAO address to the server in order to realize MLAT by TDOA.

Thank you

mhostetter, 2019-07-23:
For performing TDOA geolocation, you will need better time accuracy. Changes to `gr-adsb` are needed for that. You should probably timestamp the signal time of arrival in the ADS-B Framer. And the framer should probably determine fine time referenced to the USRP time tags (from the USRP's GPSDO) and not UNIX system time. You will need microsecond, or nanosecond, time accuracy to get good geolocation.

Tyoluyi, 2019-07-23:
Thank you for your instruction, I'll carry out the change. Moreover, could I ask whether this ZMP PUB message sink can be used for message transmission like transmit from this computer to another computer?  Sorry for too many problems. I'm a green hand. Thank you

mhostetter, 2019-07-23:
Yes, you can PUB/SUB across machines. You should set the ZMQ PUB Message Sink block's Address field to `tcp://*:5001`. Let me know if that doesn't work.

mhostetter, 2019-07-25:
I added this feature (with time referenced to the flowgraph's initial Unix timestamp) in #26. It's not nanosecond accurate, but is significantly better than before. Please check it out and let me know your thoughts.

Tyoluyi, 2019-07-26:
Thank you. I have a question. Why don't we tag a timestamp in framer.py but
in demod.py?

mhostetter, 2019-07-26:
You're right. That's probably a better place to put it. In either block, the way I have it implemented currently is the start time of the flowgraph is saved (from current UNIX timestamp) and then the burst time is determined by the SOB (start of burst) sample, at the given sample rate. Both blocks should produce the same answer. I guess I put it in `demod.py` because that is where the demodulated burst metadata is produced and the PDU is created. `framer.py` just adds a SOB tag to the sample stream.

closed 2021-09-10 by mhostetter