mhostetter/gr-adsb #25: Get Web Server to run in Python 3 with GNURadio 3.8 packages

potto216, 2019-07-24:
I would be interested in getting the Web Server to run in Python 3 with the GNURadio 3.8 packages. It seems like this could be accomplished independent of the demodulator conversion, by writing a Python program to simulate the serialization of the pmt key/values and pushing them over the ZMQ socket to the web server. This test program would also serve as a way to validate the functionality of the web server. Matt, what are your thoughts on this?

mhostetter, 2019-07-24:
The webserver language (or Python version) is completely independent of the flowgraph (3.7 vs 3.8) or the webpage operation. So it can be a Python 2 or 3 Flask application or a node application. The flowgraph-to-webserver interface is ZMQ PUB/SUB and the webserver-to-webpage interface is web sockets. As long as the Python 3 conversion supports those, which it will, it will be interoperable. So there isn't a need for a PMT serializer/deserializer program.

I do agree on the utility of recording and playing back data, however. That was actually the intent of my other project [gr-sqlite](https://github.com/mhostetter/gr-sqlite). If you connect a SQLite Sink to the ADS-B Decoder output that will log those packets to a database. You can create a separate flowgraph that reads back those packets (in time order) and transmits them to the webserver. That flowgraph would be a SQLite Timed Source connected to the ZMQ Pub Message Sink. I could easily create an example and add it the the `examples/` directory.

potto216, 2019-07-25:
If you don't mind adding an example that would be great. Sounds like gr-sqlite would be the best way to test.

mhostetter, 2019-07-25:
Yeah, I will. It will probably be next week, though.

mhostetter, 2019-07-25:
@potto216 this feature is now merged in. Let me know if it works for you and your thoughts.

potto216, 2019-07-27:
gr-sqlite is a really cool set of components. Thanks for writing it. I was able to run the graph and it successfully transmitted locations to the webserver. The webpage was able to display the location. Now that a consistent way of generating test vectors is done should the webserver be tested with Python 3 since Python 2 support ends in December.

closed 2021-12-29 by potto216

mhostetter, 2021-12-29:
@potto216 sorry for not getting back sooner. Was this closed because it was no longer needed or because the webserver did work with Python 3?

potto216, 2021-12-30:
Hi Matt, sorry for not documenting the reason for closing it. It was because it was no longer needed.