mhostetter/gr-adsb #32: timestamp and failed

sdradsb, 2020-01-15:
Hi ,
I am a graduate student and I am using gr-adsb to capture the adsb signals but I am facing some trouble. If you could guide me on this it would be really helpful.

When i used gr-adsb this is the verbose that I got:

Gtk-Message: 12:27:46.224: Failed to load module "canberra-gtk-module"
[INFO] [UHD] linux; GNU C++ version 7.4.0; Boost_106501; UHD_3.14.1.HEAD-0-gbfb9c1c7
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] Found an internal GPSDO: GPSTCXO, Firmware Rev 0.929b
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Asking for clock rate 32.000000 MHz... 
[INFO] [B200] Actually got clock rate 32.000000 MHz.
OOOOO

----------------------------------------------------------------------
Datetime:       2020-01-15 18:27:50.463353 UTC
SNR:            14.45 dB
DF:             17 Extended Squitter
CRC:            Passed
CA:             5 Level 2 or Above Transponder, Can Set CA 7, In Air
AA:             a82298
TC:             4 Identification (Category Set A)
Callsign:       NKS250  
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

----------------------------------------------------------------------
Datetime:       2020-01-15 18:27:52.824349 UTC
SNR:            12.14 dB
DF:             17 Extended Squitter
CRC:            Passed
CA:             5 Level 2 or Above Transponder, Can Set CA 7, In Air
AA:             ac491d
TC:             29 Reserved
To be implemented
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

----------------------------------------------------------------------
Datetime:       2020-01-15 18:27:54.682372 UTC
SNR:            6.92 dB
DF:             17 Extended Squitter
CRC:            Failed (PI-CRC = -858354)
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

----------------------------------------------------------------------
**Datetime:       2020-01-15 18:27:59.115507 UTC
SNR:            5.36 dB
DF:             17 Extended Squitter
CRC:            **Failed** (PI-CRC = -3883480)
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO**

----------------------------------------------------------------------
Datetime:       2020-01-15 18:28:00.954059 UTC
SNR:            14.87 dB
DF:             17 Extended Squitter
CRC:            Passed
CA:             5 Level 2 or Above Transponder, Can Set CA 7, In Air
AA:             a82298
TC:             11 Airborne Position
valid_lat_lon False
lat_cpr 92136
lon_cpr 58629
lat nan
lon nan
SS:             0 No Condition Information
Time:           0 Not Synced to 0.2s UTC Epoch
Latitude:       N/A N
Longitude:      N/A E
Altitude:       34000 ft
OOOOOOOOOOO

*In the above output it shows that some messages have been failed to display. I still am unable to figure out why I get failed messages. 

*I have an internal GPSDO installed so is the timestamp displayed the timestamp aligned with the GPSDO.

*For the last message, it is shown that time is not synced and it also doesn't display the corresponding latitude and longitude.

Thank you

mhostetter, 2020-01-27:
Sorry for the late reply.

1. The messages that faile to decode with something like `CRC: Failed (PI-CRC = -3883480)` are due to low SNR. That is to be expected.
1. The `Datetime: 2020-01-15 18:28:00.954059 UTC` fields are from system time, not the GPSDO time. You would need to modify the framer to grab the USRP stream tags to correlate GPS time and absolute sample offset. That could then be used to convert frame start-of-burst sample into true GPS time. That shouldn't be too hard, and I'm open to pull requests 😃 
1. The time syncing issue you are referring is not an issue. The `Time: 0 Not Synced to 0.2s UTC Epoch` output is decoded from the ADS-B packet. It means that the ADS-B timestamps from the ADS-B hardware are not synced to 0.2s UTC time.
1. The `OOOOOOOOOO` outputs you are getting are from overruns from your USRP hardware. This will likely cause some packets to get decoded incorrectly. Ideally, run you USRP at the lowest rate possible, with 2 Msps being the lowest.

sdradsb, 2020-02-05:
Hi, 
Thank you for the reply. I really appreciate it. I was able to fix it. I am stuck on another one though!! I am trying to get the I/Q data of the ADS-B data. I would really appreciate if you could pinpoint where in the code I could collect the I/Q data. Thanks a lot for the help.

closed 2021-09-10 by mhostetter