mhostetter/gr-adsb #33: Installation via Pybombs results in missing ADS-B Demodulator

edersonrmachado, 2020-08-11:
Hi  Matt H., thanks for this great project. I have a question about an issue that happens, ( maybe only in my `gr-adsb` installation), with  ADS-B Demodulator block. When I've installed `gr -adsb` with traditional build/make process the installation works OK, and all ADS-B blocks are listed in GRC library tree, but when I've installed with PyBOMBS, (command: `pybombs install gr-adsb`), ADS-B Demodulator is missing. 

Do you have any idea what may be happening ?

My system configuration is Ubuntu 18.04 with GRC 3.7.
I've also installed  `gr-adsb` in a docker container via Pybombs with ubuntu 18.04 and GRC 3.7 and same problems ocurred.

Here is an image of grc:

Thank you for your attention

[image]
][logo]

mhostetter, 2020-08-17:
Sorry for the late reply. My `gr-adsb` module is not on PyBOMBS. That name was taken by another project before me 😢. That's why the PyBOMBS install didn't work as expected. I'd like to get on PyBOMBS, but https://github.com/wnagele/gr-adsb beat me to it. And now his project is archived...

Instead, you'll need to do a [source build](https://github.com/mhostetter/gr-adsb#source-build) for my project. Let me know if you have trouble.

edersonrmachado, 2020-08-17:
Thanks Matt. I got confused with the names and made this wrong association seeing the list in CGRAN, sorry. The module works fine doing [source build][source build], as indicated! I used it to test an ADS-B packet transmission block with an USRP and, after some adjusts (threshold and gains) to adapt my signal generated, I could see the expected "brief" output. Thanks!!! 

[image]
/home/aqualtune/Pictures/screenshot.png")

[source build]:<https://github.com/mhostetter/gr-adsb#source-build>

closed 2020-08-17 by edersonrmachado

mhostetter, 2020-08-17:
Excellent! Glad to see it worked! 😄

edersonrmachado, 2020-08-18:
Thanks !! Sorry to directly close the issue on git, I'm learning how
discussions work and it was my first question. Your project has saved me
time. I'm trying to use docker containers to automate RF services. In that
way I'd put GNURadio and our OOT modules in a docker container and  run
the  python file corresponding to flowgraph, changing inputs like TX Gain,
Rx Address to parameters,  and it works with brief output! Now I'm going
to test the web map output of your project and see how I could direct it to
a remote user. Meanwhile I'm trying to advance my knowledge in programming
(because I'm still a beginner)

I would like to answer you about issue #27
<https://github.com/mhostetter/gr-adsb/issues/27>. Due to the Google
API-Key restrictions I noted that you've started to migrate the web part to
leaflet. It sounds great!!!
So I would like to know if this modification is already implemented, and
how I could access it.

I'm using python 2.7 and GRC main-3.7

Thanks for helping me Matt. I can keep you informed about the progress if
you're interested!

😀