mhostetter/gr-adsb #34: Google maps API key missing and cross origin requests blocked

rapbando, 2020-08-20:
HI,

when I load the tracking page from the webserver I get this errors:

[image]

[image]

mhostetter, 2020-09-03:
Hmm... You have internet access and can access maps.google.com normally? To be honest, my web knowledge is weak and I barely scrapped this webpage together. It might be difficult for me to debug. Is it still not working for you?

rapbando, 2020-09-03:
Yes connection and google maps is working, can you verify if your setup is still working with current browsers?

edersonrmachado, 2020-11-03:
You can try to change `webserver.py` line:
```python
34   socketio = SocketIO(app)
```
for this:
```python
34  socketio = SocketIO(app, cors_allowed_origins='http://localhost:5000')
```
You can also instruct the server to allow all origins with `'*'` instead of `'http://localhost:5000'`, but you must do this carefully, as stated in the documentation : <https://flask-socketio.readthedocs.io/en/latest/> 

"_A special value of '*' can be used to instruct the server to allow all origins, but this should be done with care, as this could make the server vulnerable to Cross-Site Request Forgery (CSRF) attacks_"

Besides that, If you are using Google I think you need an API key : <https://developers.google.com/maps/documentation/embed/get-api-key>

rapbando, 2020-11-03:
Thanks, It fixed the cross origin related error.
Now I´m not sure where should the API be inserted, do I have to add in the static page html code?

rapbando, 2020-11-03:
By reading [here](https://developers.google.com/maps/documentation/javascript/v2tov3) looks like the developer has to change the api version rather than adding the key or both.

edersonrmachado, 2020-11-07:
Hi! Sorry, I don't know how it works now... I'm not using Google Maps because this part of getting/use API key is very confusing. I reinstalled `gr-adsb` from `maint-3.7` branch which uses open-source `leaflet.js` (thanks to mhostetter <https://github.com/mhostetter/gr-adsb/issues/27>).

rapbando, 2020-11-08:
> Hi! Sorry, I don't know how it works now... I'm not using Google Maps because this part of getting/use API key is very confusing. I reinstalled `gr-adsb` from `maint-3.7` branch which uses open-source `leaflet.js` (thanks to mhostetter #27).

closed 2020-11-08 by rapbando

mhostetter, 2020-11-16:
Sorry for the late response.

@edersonrmachado, yes, I migrated to `leaflet.js` because a previous user noticed that the "free" version of Google Maps was no longer working as intended. I thought I had `master` and `maint-3.7` up-to-date. I will do that so others who `git pull` will get the latest 3.7 code. I'll also put a note about pulling from the `maint-3.7` or `maint-3.8` branches.

Thanks for pointing this out, and poking around enough to find the leaflet implementation! Please let me know if you encounter issues with it or have other feature requests.

edersonrmachado, 2020-11-16:
np @mhostetter, thanks for developing the leaflet version and makes your project open source