mhostetter/gr-adsb #35: ADS-B test data and FMCOMMSX usage

felixei, 2020-10-02:
Hi there, 
can you provide by any chance some pre-captured test data that works with your example flowgraph? I was trying to generate some myself, but couldn't really get there.

I am currently evaluating using your library with an FMCOMMS5 board and some test data would be highly useful for comparison (and especially some Hardware in the Loop evaluation).

In that regard, do you have any experience using different SDR baords / gr-interfaces? Like for example Analog Devices FMCOMMSX boards (using an gr-iio source block). Technically, I would assume any SDR should work as long as it provides complex float data to the flowgraph.

Thanks in advance!
Felix

felixei, 2021-02-03:
I just realised that this question was still open. I have solved it long time ago already.

For anyone who might stumble upon this with similar issues: 
My problem was to scale my test data values properly so that the iio interface sends out amplitudes that are high enough to be recognized by the gr-iio framer block. Another issue might be the selection of appropriate Sample rates in the FMCOMMSX interface (i.e. gr-iio block, or any other iio interface with control over the receive and trnasmit channels): For me, 4 MSPS for TX and RX worked just fine. Also consider watching your signals in the graphical plot (output from decoder block) to see if the threshold value has to be adjusted. 

(**Suggestion to the author**: an automatic detection threshold would be awesome, especially when this flowgraph is run on a machine without graphical GRC capability and/or human interaction. like embedded or a server.

Felix

closed 2021-02-03 by felixei

mhostetter, 2021-02-06:
@felixei your suggestion is a good one, thanks! I'll try to incorporate that. I definitely agree it's impossible to use without a GUI and various unknown gain stages -- it'll either never trigger or always trigger. An AGC or a detection threshold based on a multiple of a running power average would be much better. Thanks.

felixei, 2021-02-07:
@mhostetter thanks for getting back to that! If you have a preliminary solution, let me know. I could use it short-term and I'd be glad to test it out. I am also building a few (matlab) test scripts to generate data and evaluate performance based on different parameters like noise, anttenuation, message frequency, etc.

I am currently evaluating proof-of concept multi-channel ads-b receivers on an embedded SDR platform and your library, so far, principally runs very well, also with a set threshold. But that might change upon introducing perturbations and especially attenuation.

Any thoughts on a quick and dirty solution? I was thinking, as you alredy calculate SNR in this block, it might be possible to use that for auto-tuning the threshold on each iteration.