mhostetter/gr-adsb #36: Make master up-to-date with maint-3.7

mhostetter, 2020-11-16:
Suggested in #34.

- [ ] Make `master` and `maint-3.7` branches up-to-date with one another
- [ ] Add section in README about pulling from `maint-3.7` or `maint-3.8` for GNU Radio 3.7 and 3.8, respectively
- [ ] Update the Google Maps image in README with an image of the leaflet implementation

alphafox02, 2020-12-11:
I’ll have to give this a try again, but I seemed to be able to get everything to work minus the map server with 3.8 and python3. Is the leaflet suppose to work with 3.8?

mhostetter, 2021-05-10:
@alphafox02  I'm sorry I missed this comment before. I believe Python 3 and GNU Radio 3.8 are supported. The `master` branch now mirrors the `maint-3.8` branch. Clone either and I believe you'll have success. Please report back on if things worked.

alphafox02, 2021-05-11:
I’ll check it out again ASAP. Thank you for taking a look.

alphafox02, 2021-05-16:
@mhostetter i gave it another try after pulling with git. I tried running the map Python file and made sure to install anything it was looking for, but it kept saying Zope.interface not found. After doing some searching I found a site that suggested installing half dozen things dealing with zope. For example, zope.proxy, zope.security etc etc.

I just went down the list until the webserver.py file stopped complaining and started. I was able to access the page on port 5000. I’ll have to give it a try with adsb running.

alphafox02, 2021-05-16:
It works, pulled up the flow graph for rx, changed the source block option to “no sink” and just used it as is with an rtlsdr I had laying close by. I probably couldn’t change the source block to rtlsdr from usrp, but it worked. I saw info coming in on the terminal. Planes on the map, all good!

mhostetter, 2021-05-16:
I'm glad it worked for you. I've never heard of `zope` or those dependencies. I wonder if it's an operating system difference. Which OS are you running?

To be honest, when first developing this module, the web-related dependencies for `flask` and `webserver.py` were a huge pain. So, sadly, I'm not too surprised. Hope things work well going forward.

alphafox02, 2021-05-16:
It was on 20.04 Lubutnu. It does seems to be working good now, I’m also trying it on the Pi 20.04 64bit build. That’s also working just fine. Thanks for making this available!

closed 2025-04-09 by mhostetter