mhostetter/gr-adsb #39: Error RTL-SDR

mauriwillig, 2021-05-06:
Hi, I'm using gr-adsb for GNU 3.8 version, and I'm getting this error:

> Executing: /usr/bin/python3 -u /home/mauricio/Desktop/adsb_rx.py
> 
> gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
> built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
> The RTL TCP server reports a R820T tuner with 29 RF and 0 IF gains.
> Traceback (most recent call last):
>   File "/home/mauricio/Desktop/adsb_rx.py", line 258, in <module>
>     main()
>   File "/home/mauricio/Desktop/adsb_rx.py", line 236, in main
>     tb = top_block_cls()
>   File "/home/mauricio/Desktop/adsb_rx.py", line 176, in __init__
>     self.adsb_decoder_0 = adsb.decoder("Extended Squitter Only", "None", "Brief")
>   File "/usr/local/lib/python3/dist-packages/adsb/decoder.py", line 284, in __init__
>     self.screen = curses.initscr()
>   File "/usr/lib/python3.8/curses/__init__.py", line 29, in initscr
>     setupterm(term=_os.environ.get("TERM", "unknown"),
> _curses.error: setupterm: could not find terminal
> Error in atexit._run_exitfuncs:
> _curses.error: must call initscr() first
> 
> >>> Done (return code 1)

mhostetter, 2021-05-06:
Thanks for the report. How did you execute `adsb_rx.py`?

From looking at the traceback, the problem is not with the RTL-SDR but with the `curses` python module. From googling, it seems that `curses` doesn't work in a python interpreter but only in terminals. So I'm guessing it should work if you open a terminal and type `python3 adsb_rx.py`.

Also, I'm curious if this is related to https://bugs.python.org/issue7597.

mauriwillig, 2021-05-10:
Hi! I'm running adsb_rx.grc (from ./examples) on GNU Radio Companion 3.8.1.0, and when I run it, I get the error I have posted. 

Yes, if I only run 'python3 adsb_rx.py' on a terminal it works, but I cannot run it directly from GRC.

mhostetter, 2021-05-10:
Ok, this should be flagged as a bug. I'm guessing the change happened sometime in GNU Radio 3.8 with how they handle the "console output" or how they invoke flowgraphs. Easy fix is rip out `curses` and use something else. Maybe there's a workaround with `curses`. I'm going to leave this issue open, but it may take me a little time to resolve this. I'm preoccupied with something else at the moment.

mauriwillig, 2021-05-10:
Ok, I just edited my last comment and added a pic of the GRC console when I run it.
Let me know if I can help you with something. 

Take your time. Thanks!

mauriwillig, 2021-05-10:
Little update: Can get it working, just running "gnuradio-companion adsb_rx.grc" in console and then hitting RUN button on GRC. Yeah... probably there is a bug with how GRC invoke flowgraphs.

peternmuller, 2025-10-02:
https://github.com/mhostetter/gr-adsb/issues/59#issuecomment-3359787001