mhostetter/gr-adsb #49: gnuradio is not detecting my dongle

e-duard2, 2022-07-05:
Hi I'm new to gnuradio and to gr-adsb, I am using it for my Bachelor's thesis but I am strgling while running it:

Executing: /usr/bin/python3 -u /home/TFG/gr-adsb/examples/adsb_rx.py

[INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107100; UHD_3.15.0.0-4
Traceback (most recent call last):
  File "/home/TFG/gr-adsb/examples/adsb_rx.py", line 267, in <module>
    main()
  File "/home/TFG/gr-adsb/examples/adsb_rx.py", line 243, in main
    tb = top_block_cls()
  File "/home/TFG/gr-adsb/examples/adsb_rx.py", line 110, in __init__
    self.uhd_usrp_source_0 = uhd.usrp_source(
  File "/usr/lib/python3/dist-packages/gnuradio/uhd/__init__.py", line 125, in constructor_interceptor
    return old_constructor(*args)
  File "/usr/lib/python3/dist-packages/gnuradio/uhd/uhd_swig.py", line 2791, in make
    return _uhd_swig.usrp_source_make(device_addr, stream_args, issue_stream_cmd_on_start)
RuntimeError: LookupError: KeyError: No devices found for ----->
Empty Device Address

>>> Done (return code 1)

Please could someone help me!!!!?

mhostetter, 2022-07-05:
The USRP Source is only compatible with Ettus Research USRP devices. You'll need to swap out that source with a compatible one for your device. What dongle do you have?

e-duard2, 2022-07-06:
I'm using a RTL-SDR.COM dongle with and RTL.2832U

mhostetter, 2022-07-06:
Check for a block called "RTL-SDR Source". If you don't have that you can install `gr-osmosdr` to use the RTL dongle with GNU Radio.

You'll need to set the center frequency to `fc` and the sample rate to `fs`. You'll also need to play with the gain settings too.

You can find tutorials online of people installing the software and using the RTL-SDR for listening to FM radio, etc.

closed 2025-04-08 by mhostetter