mhostetter/gr-adsb #50: `RuntimeWarning: divide by zero encountered in divide`

wucke13, 2022-12-08:
This line yields the error. This was also mentioned in https://github.com/mhostetter/gr-adsb/issues/31. I see the error every time when I launch the script. I'm using a PlutoSDR as source.

https://github.com/mhostetter/gr-adsb/blob/5d0361d208072574e1a3a200fca07c0b334f16d0/python/demod.py#L101

mhostetter, 2022-12-08:
Are you using real captured signals or synthetically-generated signals? This error occurs if the bit 0 positions have exactly 0.0000 amplitude, which is unlikely in real captured signals. An easy fix is to add a complex noise source (with very very small noise) to your signal.

But I agree, this error should be avoided.

wucke13, 2022-12-08:
This is whenever I start gnuradio with a real SDR. Maybe the PlutoSDR sends
a bunch of zeros before actually receiving in the brief moment after the
configuration was pushed via libiio?

thinkski, 2023-03-16:
I encountered same and fixed in #52.

mhostetter, 2025-04-09:
Fixed in #52.

closed 2025-04-09 by mhostetter