mhostetter/gr-adsb #51: No GUI Mode

tkotonkla, 2023-01-10:
Could I use the ADS-B demodulator block without the QT GUI Time sink or another block? 
Because I need to run this file in No GUI Mode via the python file in raspberry pi in no desktop mode.

mhostetter, 2023-01-10:
Yes, you can. Just disable the QT GUI blocks and set the flowgraph to No GUI Mode.

tkotonkla, 2023-01-11:
Thank you so much for your response, sir.
When I follow your suggestion, The ADS-B demodulator block changed to a red block. I think it requires some output after I disable the QT GUI Time Sink block. How I fix it?

mhostetter, 2023-01-11:
Ah, yes. You'll need to connect the orange output that currently goes to the QT GUI Time Sink to a Null Sink. See here: https://wiki.gnuradio.org/index.php/Null_Sink.

tkotonkla, 2023-01-12:
Thank you, sir.
It working.
I have another question about ZMQ Pub. I try to write a python script file to get the raw data but it doesn't work.
Could you advise on how to retrieve the data?

mhostetter, 2023-01-12:
Please refer to the `webserver.py` implementation. It shows how to setup a ZMQ subscriber and receive the decoded ADS-B packets.

https://github.com/mhostetter/gr-adsb/blob/5d0361d208072574e1a3a200fca07c0b334f16d0/web/webserver.py#L44-L55

tkotonkla, 2023-01-13:
Awesome, I have done everything. Thank you, sir.

closed 2023-01-13 by mhostetter