mhostetter/gr-adsb #53: 'adsb' has no attribute 'framer'

jpreister4, 2023-04-26:
Hey, 

My group and I tried cloning the repo according to the directions and the following error is occuring:

`line 176, in __init__
    self.adsb_framer_1 = adsb.framer(fs, threshold)
AttributeError: module 'adsb' has no attribute 'framer'`

I was looking at a few of the other issues(specifically #17) and originally thought it may have been a directory issue however I'm pretty sure it isn't now. Any advice?

Thank you

mhostetter, 2023-04-26:
What version of GNU Radio are you using? How did you install `gr-adsb`?

jpreister4, 2023-04-26:
GNU Radio Version is 3.8.5.0 

[image]

How installed: 

I followed the directions again just to make sure but I did:

1) I created a new folder called "new gr-adsb"
2) went into the folder and cloned the gnuradio version 3.8 of the project
3) went into the folder "gr-absd"
4) did "mkdir build"
5) "cd build"
6) "cmake ../"
7) "make"
8) "sudo make install"
9) "sudo lpconfig"

ran it again and got the same error

Thanks again

mhostetter, 2025-04-08:
Did you ever find the root cause?