mhostetter/gr-adsb #59: Error

blackroom-0, 2023-11-12:
Using HackRF One with firmware 2021.03.1
Traceback (most recent call last):
  File "/root/gr-adsb/examples/adsb_rx.py", line 255, in <module>
    main()
  File "/root/gr-adsb/examples/adsb_rx.py", line 233, in main
    tb = top_block_cls()
         ^^^^^^^^^^^^^^^
  File "/root/gr-adsb/examples/adsb_rx.py", line 172, in __init__
    self.adsb_decoder_0 = adsb.decoder("Extended Squitter Only", "None", "Brief")
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/adsb/decoder.py", line 292, in __init__
    self.screen = curses.initscr()
                  ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/curses/__init__.py", line 29, in initscr
    setupterm(term=_os.environ.get("TERM", "unknown"),
_curses.error: setupterm: could not find terminal
Exception ignored in atexit callback: <built-in function endwin>
_curses.error: must call initscr() first

>>> Done (return code 1)

mhostetter, 2023-11-12:
What branch are you building? What version of GNU Radio? What is your OS?

It seems your OS is headless / doesn't have a terminal? Is that correct?

bmurray, 2024-06-13:
This appears to be when GnuRadio is run from a shortcut instead of from a terminal. Running GRC from the terminal seems to solve this. But that begs the question, why is it a hard dependency to need curses if people may want to run it headless? Is there a way to run it without curses?