mhostetter/gr-adsb #60: Curious output of unknown origin

JBKingdon, 2023-11-13:
Hi, I'm new to gnu radio so I apologise if this is a misguided question.

I've been looking at gr-adsb as a learning exercise, using a hackrf as the sdr. I'm seeing very occasional reports in the terminal, but also the divide by zero error discussed in https://github.com/mhostetter/gr-adsb/issues/50 and some strange (debug?) output that I haven't tracked down yet. This appears differently in the linux terminal vs the gnu radio companion console (if that's the right term for the bottom left region of the UI).

In companion I see a mixture of 

`OOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsOOsO
`
and

`[3d14:55:33 c03f81[3;27H1825[77G1[7G47[78G[3;36H0   220    97[77G2[8G8[78G[8G9[3;33H-192[3;48H5[77G4[7G51[3;33H  
`
while in the linux terminal I only see the OOs sequences.

Are these significant or can I safely ignore them?

mhostetter, 2023-11-13:
The OOOs are likely overflows from the SDR. So it is producing more samples than the receiver can process. What is the HackRF's sample rate? I would consider halving it and seeing if the overflows disappear.

The divide by zero error is a nuisance, not really a problem.

JBKingdon, 2023-11-13:
Thanks for the lightning fast response!
The hackrf samples at up to 20M, I currently have fs set at 4M. Processor is i7-6820HQ

closed 2025-04-08 by mhostetter