mhostetter/gr-adsb #62: Which version of flash-socket to use for Webserver?

caporalfourrier, 2024-03-10:
I am getting the following issue when I open the webserver:

`The client is using an unsupported version of the Socket.IO or Engine.IO protocols (further occurrences of this error will be logged with level INFO)`

I can open the google maps page but I see no planes even though I am continuously receiving data. No issues with ZMQ either. I tried downgrading to a few different older versions but it didn't work and now the webserver.py itself won't run. So what version should I use?

Malocch1o, 2024-04-02:
I'm encountering the same error and the same issue with the map not showing planes despite receiving data. Have you worked out a fix for this? @caporalfourrier
Perhaps @mhostetter can provide some insight.

iMRUM, 2025-04-09:
Note that client-side code is using Socket.IO version 2.2.0 (line 40 in index.html):
```
<script src="https://cdnjs.cloudflare.com/ajax/libs/socket.io/2.2.0/socket.io.js"></script>
```
 Perhaps it has something to do with your Socket.IO version.

Check both flask-socketio and python-socketio to make sure they're aligned by running:
```
pip show python-socketio
pip show flask-socketio
```

If you have a newer version, you need to update index.html.
For instance, for Socket.IO server 4.x or 5.x, modify line 40 as follows (works with python-socketio v5.12.1 and Flask-SocketIO v5.5.1):
```
<!-- Replace this line -->
<script src="https://cdnjs.cloudflare.com/ajax/libs/socket.io/2.2.0/socket.io.js"></script>

<!-- With this line -->
<script src="https://cdnjs.cloudflare.com/ajax/libs/socket.io/4.8.1/socket.io.js"></script>
```

I have encountered the same problem and fixed it in my fork:
https://github.com/iMRUM/gr-adsb

Hope this helps!